msm8916-mainline/lk2nd #567: msm8909 (Nokia 6300 4G): only single cpu core is started

mo-krauti, 2025-08-04:
lk2nd: v20.0
pmOS: edge

journalctl reports:
```
kernel: Failed to set CPU boot address, disabling SMP
kernel: Setting up static identity map for 0x80100000 - 0x80100060
kernel: rcu: Hierarchical SRCU implementation.
kernel: rcu:         Max phase no-delay instances is 1000.
kernel: Timer migration: 1 hierarchy levels; 8 children per group; 1 crossnode level
kernel: EFI services will not be available.
kernel: smp: Bringing up secondary CPUs ...
kernel: smp: Brought up 1 node, 1 CPU
kernel: SMP: Total of 1 processors activated (38.00 BogoMIPS).
kernel: CPU: All CPU(s) started in SVC mode.
```
lscpu:
```
Architecture:             armv7l
  Byte Order:             Little Endian
CPU(s):                   1
  On-line CPU(s) list:    0
Vendor ID:                0x41
  Model name:             ARMv7 Processor rev 5 (v7l)
    Model:                5
    Thread(s) per core:   1
    Core(s) per socket:   1
    Socket(s):            1
    Stepping:             0x0
    BogoMIPS:             38.00
    Flags:                half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
```
htop also only shows one core.

wonderfulShrineMaidenOfParadise, 2025-08-05:
> [    0.069816] Failed to set CPU boot address, disabling SMP

@V6lhost also reported similar issue
https://linux-hardware.org/?probe=379cf72f84&log=dmesg

But I am not able to reproduce it.

> [    0.106036] smp: Brought up 1 node, 4 CPUs

https://linux-hardware.org/?probe=0ec8c8c45f&log=dmesg

mo-krauti, 2025-08-06:
My hw-probe: https://linux-hardware.org/?probe=d7a4b1d49b

And the lk2nd boot log (I know images are not optimal but I do not have a UART connection right now):

[image]